LightningPiggy/lightning-piggy #7: Show on-chain balance

dadofsambonzuki, 2023-12-07:
It would be great for the kids to show their on-chain balance too.

Lightning = spends.
On-chain = savings.

The LNBits `Onchain Wallet` extension enables this after sharing the wallet XPUB.

dadofsambonzuki, 2023-12-07:
`/watchonly/api/v1/wallet/{wallet_id}` returns balance apparently, but my instance returns zero.

ThomasFarstrike, 2024-01-31:
I like this idea, we could do it as part of the "multiple screens" update.

So the proposal is to add a new screen that shows something like:
```
On-chain:  XXXXXXXX sats
Off-chain:    YYYYY sats
------------------------
Total:    ZZZZZZZZZ sats
```
To show the 2 amount with the sum aligned below makes it clear that they're distinct and is also a learning opportunity for the child to make the sum themselves and check that the digits add up correctly.

To be determined how exactly we show these big sat amounts like which separator (space, comma, dot, none) which unit (sats/BTC/bits).

Optional: also show an on-chain receiving QR code so it's easy to send (bigger amounts of bitcoin) to the on-chain wallet. To be determined (or allowing configuring) whether to reuse the same UTXO (tradeoff: privacy), or show a new one every time (tradeoff: small UTXO's so higher fees), or to show a new one every time the same UTXO has reached a threshold amount (tradeoff: coding complexity).