LightningPiggy/lightning-piggy #12: Feature Request: More clear battery indicator

BenGWeeks, 2024-05-15:
Most users won't know the device doesn't have enough power at 3.3V (or there abouts) and an indicator to that explain the battery is dangerously low and might not switch on again.

ThomasFarstrike, 2024-05-19:
Yeah, true. I guess a % would be the most clear, like we the wifi signal. Minimum is 3.7V and max is 4.2V.

So 4.2V should show 100%, 3.7V should show 0% and 3.95V would 50%.

ThomasFarstrike, 2024-05-19:
Another idea is to show a battery icon with a few bars, but that's more difficult to implement (needs some custom drawings) and also less precise so I would keep it simple and precise for now by using a %, which can then be converted back to a voltage for anyone who requires it.

ThomasFarstrike, 2024-05-19:
Done, see commit cba095bc109f87b2a81d6bbd96a39078eb631843

closed 2024-05-19 by ThomasFarstrike