LightningPiggy/lightning-piggy #13: Feature Request: "Shake to Refresh"

BenGWeeks, 2024-05-15:
Rather than a "Shake-to-wake", it would be good if this was a "Shake-to-refresh" which makes it more clear if your battery is dead or not as well.

ThomasFarstrike, 2024-05-19:
I like this idea, and I think the plan was indeed to add some functionality to the tilt sensor input pin, even when the device is already awake.

To move in this direction, I reworked the code to do periodical (every 15 minutes) refreshes if it stays on.

Next step would be to implement an interrupt handler for the tilt sensor input pin (see: https://deepbluembedded.com/esp32-external-interrupts-pins-arduino-examples/). Then in that interrupt handler, set some variable like forceBalanceAndPaymentsRefresh to "true", add "forceBalanceAndPaymentsRefresh==true" to the refresh check in the loop() and don't forget to set forceBalanceAndPaymentsRefresh=false to mark it as done.

ThomasFarstrike, 2024-08-05:
I think this is done, I forgot to close this issue!

closed 2024-08-05 by ThomasFarstrike