LightningPiggy/lightning-piggy #16: Feature Request: Add "shake-to-reset" and battery indicator to Setup "Splash"

BenGWeeks, 2024-05-15:
Because we want to ensure the boards are working correctly, it would be good to have the ability to confirm the "shake-to-reset" and the battery are working correctly even with the "Setup Splash" firmware.

ThomasFarstrike, 2025-03-07:
Now that #21 is done and the default firmware works out of the box, and is self-contained with the configuration page, I don't see the need for a separate "Setup Splash" firmware anymore. You can just install the default firmware, without configuration, and then at boot the device will guide the user to connect to its access point and open the configuration page in the webbrowser.

@BenGWeeks is it okay if I retire the old "Setup Splash" firmware so that we have just one-size-fits-all to maintain? Thanks!

ThomasFarstrike, 2025-03-28:
Setup Splash is deprecated because the default firmware works out of the box, and is self-contained with the configuration page.

You can just install the default firmware, without configuration, and then at boot the device will guide the user to connect to its access point and open the configuration page in the webbrowser.

closed 2025-03-28 by ThomasFarstrike