LightningPiggy/lightning-piggy #17: Feature Req: be able to define a different lnbits port

editwentyone, 2024-05-24:
I use my own lnbits made with raspibolt.org guide.
its sitting with duckdns on port 4003 and goes to nginx reverse port 5000 internally.

hope we can have a field to define a custom port if needed that can fallback to 443 if not set. 

thank you in advance :)

ThomasFarstrike, 2024-06-01:
Done! It's available as a configuration "LNBits Port" in the webinstaller and in the source code.

Feel free to send some sats my way, I'm at tfar@getalby.com . Much appreciated!

closed 2024-06-01 by ThomasFarstrike

editwentyone, 2024-06-01:
chatgpts helped to add the custom port just to lnbitsHost

```String getEndpointData(const char * host, String endpointUrl, bool sendApiKey) {
  int connectionAttempts = 0;

  int port;
  if (strcmp(host, lnbitsHost) == 0) {
    port = getConfigValueAsInt((char*)lnbitsPort, DEFAULT_LNBITS_PORT); 
  } else {
    port = 443; 
  }

  Serial.println("Fetching " + endpointUrl + " from " + String(host) + " on port " + String(port));

  WiFiClientSecure client;
  client.setInsecure();   https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/README.md
  client.setHandshakeTimeout(HTTPS_TIMEOUT_SECONDS);
  client.setTimeout(HTTPS_TIMEOUT_SECONDS * 1000);

  while (feed_watchdog() && connectionAttempts < MAX_HTTPS_CONNECTION_ATTEMPTS && !client.connect(host, port)) {
      connectionAttempts++;
      Serial.println("Couldn't connect to " + String(host) + " on port " + String(port) + " (attempt " + String(connectionAttempts) + "/" + String(MAX_HTTPS_CONNECTION_ATTEMPTS) + ")");
  }
  if (!client.connected()) return "";

  String request = "GET " + endpointUrl + " HTTP/1.1\r\n" +
               "Host: " + String(host) + "\r\n" +
               "User-Agent: " + getFullVersion() + "\r\n" +
               "Content-Type: application/json\r\n" +
               "Connection: close\r\n";
  if (sendApiKey) request += "X-Api-Key: " + String(invoiceKey) + "\r\n";
  request += "\r\n";

  feed_watchdog();
  client.print(request);

  long maxTime = millis() + HTTPS_TIMEOUT_SECONDS * 1000;

  int chunked = 0;
  String line = "";
  while (client.connected() && (millis() < maxTime))
  {
    line = client.readStringUntil('\n');
    line.toLowerCase();
    if (line == "\r")
    {
      break;
    } else if (line == "transfer-encoding: chunked\r") {
      Serial.println("HTTP chunking enabled");
      chunked = 1;
    }
  }

  if (chunked == 0) {
    line = client.readString();
  } else {
    
    String lengthline = client.readStringUntil('\n');
    Serial.println("chunked reader got length line: '" + lengthline + "'");

    line = "";
    while (lengthline != "0\r" && (millis() < maxTime)) {
      const char *lengthLineChar = lengthline.c_str();
      int bytesToRead = strtol(lengthLineChar, NULL, 16);
      Serial.println("bytesToRead = " + String(bytesToRead));

      int bytesRead = 0;
      while (bytesRead < bytesToRead && (millis() < maxTime)) { // stop if less than max bytes are read
        uint8_t buff[BUFFSIZE] = {0}; // zero initialize buffer to have 0x00 at the end
        int readNow = min(bytesToRead - bytesRead,BUFFSIZE-1); // leave one byte for the 0x00 at the end
        //Serial.println("Reading bytes: " + String(readNow));
        int thisBytesRead = client.read(buff, readNow);
        //Serial.println("thisBytesRead = " + String(thisBytesRead));
        if (thisBytesRead > 0) {
          bytesRead += thisBytesRead;
          String stringBuff = (char*)buff;
          line += stringBuff;
        } else {
          Serial.println("No bytes available from HTTPS, waiting a bit...");
          delay(42);
        }
        //Serial.println("chunked total reply = '" + reply + "'");
      }

      // skip \r\n
      client.read();
      client.read();

      // next chunk length
      lengthline = client.readStringUntil('\n');
      Serial.println("chunked reader got length line: '" + lengthline + "'");
    }

  }
  //Serial.println("returning total chunked reply = '" + reply + "'");
  feed_watchdog(); // after successfully completing this long-running and potentially hanging operation, it's a good time to feed the watchdog
  return line;
}```