LightningPiggy/lightning-piggy #23: NWC for Alby Hub etc.

Mannohat, 2024-08-13:
We discussed the possibility for an easy way to use NWC for connecting to e.g. Alby Hub which is an open-source, self-custodial wallet, with an integrated lightning node, where users hold their own keys.

Questions:
A: Is this feasible? 
B: Would it still allow for attaching a note to payments?
C: Would using the Alby App store with a designated Lightning Piggy connection allow for easier setup or any other benefits?

ThomasFarstrike, 2025-01-01:
Nostrduino for ESP32 seems to support NWC: https://github.com/riccardobl/nostrduino/

N1coB, 2025-01-30:
NWC would be great! Because that solves the Problem, that most of the useres doesn't run a clearnet node with a working lightning address etc.

With AlbyHub connected to an Alby Account you can have a working LN-Address. And with AlbyHub "Subaccounts" you can create a separated Wallet for the Piggy connected via NWC.

At the end, the Lightning Piggy App would be listed at the AlbyHub App-Store which will help for discovery.

ThomasFarstrike, 2025-01-31:
> A: Is this feasible?

I've been looking into this, doing lots of tests, getting to know NWC hands-on and it seems to be a viable option!

Mind you that Nostr Wallet Connect is still relatively new, and not many wallets support it, but at least a few do. The best one at the moment seems to be Alby, either through the "legacy" hosted wallet or the new "Alby Hub".

> B: Would it still allow for attaching a note to payments?

Yes. The idea would be that the "getbalance" and "list payments" functionality of the piggy would use Nostr Wallet Connect to fetch the info from a wallet. For receiving payments, including the payment amount and comment/memo/note, the piggy would show a QR code, with either a lightning address or an LNURLp (or even a BOLT12 in the future) that routes the payment to the (same) wallet.

Not all lightning address implementations currently support adding a custom memo/note to it. npub.cash, for instance, doesn't have the "commentAllowed" in the .well-known/lnurlp reply, making it impossible to add a comment to the payment. Others, such as coinos.io, just put a fixed comment "payment for yourname@coinos.io" in there. But at least Alby supports it. I also heard that LifPay supports it.

***

The general advantage is that NWC is an open **protocol** for connecting to any wallet that supports it, whereas LNBits (on which the current architecture is based) is a specific **software** implementation of a wallet with an implementation-specific API.

Adding support for NWC would allow the piggy to work with multiple different wallet implementations, rather than being bound to a specific one.

ThomasFarstrike, 2025-03-07:
Meanwhile, this is done, in release 6.x on the nostr-wallet-connect branch!

Also, coinos.io added support for custom payment description (also known as memo, comment, note) so that works too.

Hopefully npub.cash will also add support for payment comments (I don't see why not) but we can already close this, as it's done!

closed 2025-03-07 by ThomasFarstrike