LightningPiggy/lightning-piggy #24: Multiscreens option

Mannohat, 2024-08-13:
A way to preview different screen layouts incl. what information and how this is presented etc.

Do you users want to:
A: Show fiat value of savings 
B: Do they want to include on-chain savings (XPUB or simply enter sats amount saved on-chain) *obs. privacy
C: Big, medium, small QR preferences
D: Quotes of the day etc.
E: Loop screen / quotes on update etc.

Perhaps a mock-up of 5 different layouts and a poll on some if we decide to just go with one?
TBD

BitcoinCaddy, 2025-02-15:
A fixed Fiat amount would be great to add the amount of my kids Fiat savings!

ThomasFarstrike, 2025-03-07:
> A fixed Fiat amount would be great to add the amount of my kids Fiat savings!

Your kids have fiat savings?! ;-)