LightningPiggy/lightning-piggy #36: Add support for UTF-8 unicode emoticons and symbols

ThomasFarstrike, 2025-03-28:
It would be nice if users could put emoticons and symbols in their payment comments like 😊, 😂, ★, 😉, 🎉 

Supporting these isn't trivial because it would involve generating and installing additional fonts, in different font sizes, without blowing up the size of the build, while we are already at the limit of our partition.

But it might be doable and it might be fun, especially for kids!

Mannohat, 2026-01-09:
This would be very nice, also in Denmark we use these latin letters with diacritics a lot: æ ø å Æ Ø Å which now shows as rectangular squares on the screen. And yes ofc emojis would be awesome 😀🔥🚀
I understand that UTF-8 support would fix both

ThomasFarstrike, 2026-01-21:
Superseded by https://github.com/MicroPythonOS/MicroPythonOS/issues/24 and https://github.com/MicroPythonOS/MicroPythonOS/issues/23

closed 2026-01-21 by ThomasFarstrike